f4pga/ideas #2: Create a parser for the XDC file format

mithro, 2017-12-30:
XDC is the new file format used for constraints in Vivado. To be able to parse designs which use XDC file format we need a parser for it.

The XDC file format is described here -> https://www.xilinx.com/support/documentation/sw_manuals/xilinx2017_2/ug903-vivado-using-constraints.pdf

XDC is based on SDC - The Synopsys Design Constraints format. So the parser here -> https://github.com/verilog-to-routing/vtr-verilog-to-routing/tree/master/libs/EXTERNAL/libsdcparse might be useful.

> There are key differences between Xilinx Design Constraints (XDC) and User Constraints File
(UCF) constraints. XDC constraints are based on the standard Synopsys Design Constraints
(SDC) format. SDC has been in use and evolving for more than 20 years, making it the most
popular and proven format for describing design constraints.

Nic30, 2018-11-24:
XDC is a subset of TCL, there is a ANTRL4 grammar for TCL. There is a grammar for ANTRL4 https://github.com/Nic30/hdlConvertor/blob/master/grammars/tcl.g4

mithro, 2018-11-24:
@Nic30 Having a grammar is the first step. Still need to process it into more useful constructs.