f4pga/ideas #4: Rewrite the xml parser in vpr to be based on DTD and auto generated.

mithro, 2017-12-30:
Currently the xml parser in vpr is hand coded. 

It would be good to rip it out and replace it with a auto-generated xml parser based around a DTD which defines the file format.

mithro, 2017-12-30:
XML formats include;
 * [FPGA Architecture Definition](https://docs.verilogtorouting.org/en/latest/arch/#fpga-architecture-description)
 * [Routing Graph File Format](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#routing-resource-graph-file-format-xml)

Laksen, 2018-07-15:
I created schemas for the packed netlist and routing resource formats here: https://gist.github.com/Laksen/dc3fdf65a7eb6bb1186c46381215ac55

They don't match the examples 100% as the examples contain some undocumented values, but that should be easy to change

mithro, 2018-07-15:
@Laksen this is awesome! Do you want to send them as a pull request in the https://github.com/SymbiFlow/symbiflow-arch-defs repo? The [common/xml](https://github.com/SymbiFlow/symbiflow-arch-defs/tree/master/common/xml) directory would be the right place.

mithro, 2018-07-16:
@Laksen has added a schema for the existing XML formats in the pull request here -> https://github.com/SymbiFlow/symbiflow-arch-defs/pull/195 - Thank you so much.

The next steps would be;
 a) Figuring out the best way to generate Python and C++ code from the schemas. 
 b) Designing a new XML format (and appropriate schema) for the placement and route formats from VtR.
 c) Using the XML schema's for automatic validation of the files.

mithro, 2018-09-19:
https://www.codesynthesis.com/products/xsd/ looks pretty good but is under a GPL license and we really want to keep things open. The alternative is probably to follow an example shown at this page https://blind.guru/bmmlcxx.html and use Python + Jinja2 templates to create C++ code.

mithro, 2018-09-19:
The best example I have found is https://github.com/mlang/xsdcxx-musicxml