f4pga/ideas #6: Add support for EDIF to Yosys

mithro, 2017-12-30:
Yosys can produce EDIF but not read EDIF. It would be nice if EDIF reading support was added to Yosys.

mithro, 2017-12-30:
https://en.wikipedia.org/wiki/EDIF

mithro, 2017-12-30:
The EDIF parser could be shared with #5

rochus-keller, 2018-12-26:
Yosys generates EDIF 2 0 0 only and as far as I can see also the big tools (Vivado etc.) only seem to support EDIF 2 0 0, but each with its own "dialect". I own the standard (EIA-548); it's nearly 500 pages long. What is the goal of Yosys EDIF import? Just reading the EDIF file created by Yosys, or data exchange with other tools (Vivado etc.)? The latter seems to be a big project (with questionable feasibility). And EDIF 3 or 4 would be yet a different league again.

peterzieba, 2023-11-06:
I came across [Spydrnet](https://github.com/byuccl/spydrnet) which can read EDIF and convert it to Verilog (as well as generally modify netlists). I've been playing with it recently and found it to work pretty with the EDIF 2 0 0 files I have... Maybe this is of interest?

Incidentally, it supports Triple modular redundancy, which might also be of interest in rad-hard applications....

mithro, 2023-11-06:
FYI - Spydrnet was developed by BYU who have also contributed heavily to F4PGA toolchain and have a whole course at https://byu-cpe.github.io/ComputingBootCamp/pages/schedule/