f4pga/ideas #7: Create a JSON schema for the Yosys JSON format.

mithro, 2017-12-30:
[Yosys can write a JSON format.](http://www.clifford.at/yosys/cmd_json.html). It would be good if there was a specification for this file format.

Write a JSON Schema for the JSON format - see http://json-schema.org/

drom, 2018-11-26:
Great idea.
I have used this format for my FPGA config visualizer.  https://github.com/drom/icedrom
I can definitely help with a schema.

mithro, 2019-05-14:
FYI [netlistsvg has written up somewhat of a json schema file here](https://github.com/nturley/netlistsvg/blob/hierarchy/lib/yosys.schema.json).

proppy, 2019-11-14:
Do you have examples of Yosys JSON output? If the set is large enough something like https://github.com/ncarlier/genjson could be used to generated a merged schema.