f4pga/ideas #10: Apply prjxray ideas to document the bitstream for Spartan 6 parts

mithro, 2018-02-12:
# [prjxray] Apply prjxray ideas to document the bitstream for Spartan 6 parts

# Brief explanation

Spartan 6 is a hugely popular part which would be awesome to have support for in SymbiFlow

## Expected results

 * Documentation of the bitstream for Spartan 6 parts.
 * Support for using Spartan 6 in the SymbiFlow (IE Verilog to Routing).

# Detailed Explanation

SymbiFlow currently only supports Xilinx Series 7 parts and the Lattice iCE40 parts. While the Spartan 6 has mostly be superseded by the Artix 7 and Spartan 7 there are still a **huge** number of boards out there with Spartan 6 parts. Due to its huge popularity it will be a long time until the part is no longer in use (people still start new designs with Spartan 3!).

To make it even more important Spartan 6 designs you still have to use ISE which is significantly worse then Vivado in many ways. Having an non-ISE toolchain for that would be **awesome**.

Spartan 6 is used heavily by a number of open source projects;
 * The Novena Open Laptop - https://www.crowdsupply.com/sutajio-kosagi/novena 
 * TimVideos HDMI2USB Project - https://hdmi2usb.tv

## Further reading

 * [Link to a PDF document](http://abc/abc.pdf)
 * [Link to a some HTML page](http://abc/abc.html)

# Knowledge Prerequisites

 * List of
 * what knowledge is required
 * to complete this project.

mithro, 2018-02-12:
* https://github.com/YosysHQ/yosys/issues/448

kc8apf, 2018-02-12:
From what I've seen (Spartan-6 FPGA Configuration User Guide UG380), low-level bitstream format (BIT header, but-width detection, synchronization word) is the same between Spartan6 and 7-series.  Packet structure and overall configuration sequence are similar in concept and design but Spartan 6 uses a 16-bit word and have 65 word frames.  16-bit word length leads to a slightly different packet wire format.  Config registers are very similar for actual config loading (writes to FAR and FDRI) while other device-control-focused config registers are very different.  I expect writing a parser that yields frame data will be fairly straightforward to anyone who reads through xc7series::BitstreamReader, xc7series::ConfigPacket, xc7series::Configuration, UG470 (7-series configuration user guide), and UG380 (Spartan-6 configuration user guide)/

pisgah-fogel, 2019-10-26:
I heard about project concisting in analysing Spartan bitstream:
- [FPGA Tools](https://github.com/Wolfgang-Spraul/fpgatools) JSON to xc6slx9 bitstream (Old and not complete)

According to [blogs...](https://vjordan.info/log/fpga/bitstream-analysis-code.html) and [UG380](https://www.xilinx.com/support/documentation/user_guides/ug380.pdf) we already have some materials to start with.

Does someone is working on Spartan6 timing, bit files analysis... in order to extend SymbiFlow project to Spartan ?
I would like to develop something like XRay for ISE that allow analysis of old FPGAs. Let me know if you want to join me and/or if you are interested.

mpictor, 2020-06-05:
> I would like to develop something like XRay for ISE that allow analysis of old FPGAs

It would be nice to see this happen! Gotta find time to help... :hourglass: 

Is there a write-up of the methodology used with 7 series chips? And/or a corpus of inputs that produces interesting outputs?

DaAwesomeP, 2021-02-08:
@mithro would you mind updating the Knowledge Prerequisites section? I would like to help out if I may, but I am really unsure what I would be able to contribute without knowing the tasks required (or what I need to learn).