f4pga/ideas #14: Package all of SymbiFlow dependencies into a conda repository

mithro, 2018-03-03:
# Package all of SymbiFlow dependencies into a conda repository

Conda is a system for deploying Python + other tools on Windows, Linux and Mac. As we use a lot of Python in SymbiFlow, it would be good to provide a complete environment through conda.

## Expected results

The following tools are package in conda;
 - [ ] Yosys 
 - [ ] Verilog To Routing's VPR
 - [ ] prjxray tools
 - [ ] icestorm tools

# Detailed Explanation

conda has been used quite successfully by the [litex-buildenv](https://github.com/timvideos/litex-buildenv) to provide toolchains for their FPGA development. You can find their conda packages here -> https://github.com/timvideos/conda-hdmi2usb-packages - They would like it packaged too. -
 https://github.com/timvideos/conda-hdmi2usb-packages/issues/12

From https://conda.io/docs/
> Conda is an open source package management system and environment management system that runs on Windows, macOS and Linux. Conda quickly installs, runs and updates packages and their dependencies. Conda easily creates, saves, loads and switches between environments on your local computer. It was created for Python programs, but it can package and distribute software for any language.

mithro, 2018-03-06:
* [Yosys is now in the TimVideos repo](https://github.com/timvideos/conda-hdmi2usb-packages/tree/master/yosys)
 * [Verilog to Routing is also in the TimVideos repo](https://github.com/timvideos/conda-hdmi2usb-packages/tree/master/vtr)

dvc94ch, 2018-05-26:
It seems that the best way currently to do this would be to create AppImages. I don't know if you are familiar with functional package management - a method for constructing linux distributions. This doesn't solve the problem for application distributors - they still need to support each and every linux distribution which is a PITA. It looks like using functional package descriptions to create an appimage that will run on any distro is the way to go IMO.

- [0] https://github.com/matthewbauer/nix-bundle