f4pga/ideas #15: Symbiflow package manager

dvc94ch, 2018-03-13:
# Symbiflow package manager / build system

# Brief explanation

To be productive writing fpga code there needs to be a great package manager. Not sure if it is in the scope of symbiflow, but using the rust project as priori art, and the fact that most toolchains without a defacto package manager have pretty terrible and fragmented solutions, it makes sense to have a package manager be tightly coupled to a toolchain.

It should be hdl agnostic - so adding preprocessing steps before verilog should be easy. This would be the basis for integrating symbiflow into existing IDE's. To quickly get a non-sucking package manager up and running an option would be to start out with a non-sucking package manager like cargo (using it as a dependency), to get the configuration file parsing, dependency install|upgrade|lock and package publishing with little work. Then adding custom commands / replacing existing commands.

Due to my limited experience (approximates non-existent), I'm need some feedback on features and design.

# Detailed Explanation
Command line interface

```sh
symbiflow (new|install|lock|build|test|sim|synth|flash|publish|netlist|floorplan)

symbiflow flash --board icestick
```

Example configuration file `Symbiflow.toml`
```toml
[package]
name = "some name"
version = "0.1.0"
authors = ["david@craven.ch"]
repository = "https://github.com/dvc94ch/example"
license = "ISC"

[dependencies]
ip = "0.1.0"
other-ip = { git = "https://github.com/owner/repo" rev="sha1" }

[build]
language = "myhdl"

[board.icestick]
target = "ice40-1k-tf144"

[target.ice40-1k-tf144]
pnr = "arachne-pnr"
```

Example project layout
```
.
├── boards
│   └── icestick
│       ├── top.pcf
│       └── top.py
├── rtl
│   └── top.py
├── Symbiflow.toml
└── tests
```

kc8apf, 2018-03-16:
Does something like FuseSoC meet your needs in this space?

dvc94ch, 2018-03-17:
What I was looking for was https://github.com/cfelton/rhea, that's exactly what I want. A shame I already wasted three days on writing my own package manager...

mithro, 2018-03-17:
While I don't want to discourage you from working on something you are finding interesting and fun, the general SymbiFlow philosophy is (where possible) to take the current best existing solution and work on improving that rather than writing stuff from scratch.

I'd encourage you to reach out to the FuseSoC developer @olofk and see if you can collaborate on that. I believe he already has a library which does exactly what you are after which is reusable outside of FuseSoC. Collaborating with @cfelton would be another awesome idea!

dvc94ch, 2018-03-17:
@mithro yes that makes sense. I'm not aware of all the stuff that is available in the fpga space. I was aware of fusesoc.

I did also write a useful tool - maybe already exists - to extract a signals from a vcd dump for protocol analysis with sigrok-cli. (fails to parse if there are any vectors in the dump)

mithro, 2018-07-14:
So from the FPGA toolchain side (rather than the HDL cores / libraries point of view) I had two thoughts;

 * After the success that @tinyfpga has had with [apio](https://github.com/FPGAwars/apio) and icestorm for this [TinyFPGA boards](https://tinyfpga.com/bx/guide.html) I think that is a good option to start with.

 * I also think that [conda](https://conda.io/docs/) is a probably a good option as we have had a lot of success with that for [LiteX-BuildEnv](https://github.com/timvideos/litex-buildenv).

It might actually be the **combination** of the two things (conda to provide an easy Python environment) and APIO on top of that Python environment.

drom, 2018-11-26:
How about using existing package manager?
I like `npm`. It is the most popular package manager in the world.
It has ton of useful libraries already.
Just pack all our EDA libraries, tools, components and designs on 'npm' ?

mithro, 2020-04-15:
Few more links;
* [EDDA - Conda based system for FPGA and ASIC Dev](https://docs.google.com/document/d/1BZcSzU-ur0J02uO5FSGHdJHYGnRfr4n4Cb7PMubXOD4/edit#heading=h.ee3kfes8c7nu) -- Binary distribution system for EDA tooling based around a popular Python based system for distribution Python binary packages.
 * [FuseSoC](https://github.com/olofk/fusesoc) (“apt get of HDL cores”)