f4pga/ideas #17: Create a really good library for parsing / producing SDF files

mithro, 2018-06-24:
# Create a really good library for parsing / producing SDF files

[Standard Delay Format (SDF) format](https://en.wikipedia.org/wiki/Standard_Delay_Format)  is an IEEE standard for the representation and interpretation of timing data for use at any stage of an electronic design process. It finds wide applicability in design flows, and forms an efficient bridge between Dynamic timing verification and Static timing analysis.

It has usually two sections: one for interconnect delays and the other for cell delays.

SDF format can be used for back-annotation as well as forward-annotation.

The SDF format version 2.1 is produced by [Verilog to Routing for post implementation timing analysis](https://docs.verilogtorouting.org/en/latest/tutorials/timing_simulation/). It is also used by a lot of other EDA tools.

## Expected results

A Python library published on PyPi which makes it easy to read and write SDF files.

## Further reading

 * [More information here.](http://vlsi.pro/standard-delay-format-sdf-file/)

# Knowledge Prerequisites

 * Python

euripedesrocha, 2018-07-15:
I can put some effort here, do you have any use cases or an idea for the interface? How the information will be consumed and produced?

mithro, 2018-07-15:
The data will mainly be used to generate delay specifications for Verilog
to Routing. See
http://vtr-verilog-to-routing.readthedocs.io/en/latest/tutorials/arch/timing_modeling/index.html
for example.

mithro, 2018-07-15:
Another common thing that you would want to do with the SDF delays it to split or join them together.

If a pathway when `A -> B -> C`, you would want to add the delay `A->B` and `B->C`. Conversely if you had delay 'A -> C' you might want to confirm that `A->B + B->C` is the same value.

Another common thing you might want to do is if you have the following set of delays;
`A->B->C = 100ps`
`A->B->D == 40ps`
`E->B->D == 20ps`
Then you want to work out what the possible range of values of each of the sections would be.

mithro, 2018-07-15:
With something like this;
```python
from collections import namedtuple

Times = namedtuple("Times", ("min", "typical", "max"))

IoPath = namedtuple("IoPath", ("src", "dst", "low2high", "high2low"))

lt_paths = [
    IoPath("in0",           "ltout", Times(293.136, 324.149, 364.700), Times(310.048, 342.850, 385.740)),
    IoPath("in1",           "ltout", Times(259.313, 286.747, 322.619), Times(304.411, 336.616, 378.727)),
    IoPath("in2",           "ltout", Times(248.039, 274.280, 308.592), Times(276.225, 305.448, 343.659)),
    IoPath("in3",           "ltout", Times(214.215, 236.878, 266.511), Times(219.852, 243.112, 273.525)),
]

lc_paths = [
    IoPath("in0",           "lcout", Times(360.783, 398.952, 448.861), Times(310.048, 342.850, 385.740)),
    IoPath("in1",           "lcout", Times(321.323, 355.317, 399.767), Times(304.411, 336.616, 378.727)),
    IoPath("in2",           "lcout", Times(304.411, 336.616, 378.727), Times(281.862, 311.682, 350.673)),
    IoPath("in3",           "lcout", Times(253.676, 280.513, 315.606), Times(231.127, 255.579, 287.552)),
]
```

We can calculate the common part of the delay between `XX->ltout` and `XX->lcout`
```
in0
            ltout       lcout       delta    
       min  293.136000  360.783000  67.647000
   typical  324.149000  398.952000  74.803000
       max  364.700000  448.861000  84.161000

in1
            ltout       lcout       delta    
       min  259.313000  321.323000  62.010000
   typical  286.747000  355.317000  68.570000
       max  322.619000  399.767000  77.148000

in2
            ltout       lcout       delta    
       min  248.039000  304.411000  56.372000
   typical  274.280000  336.616000  62.336000
       max  308.592000  378.727000  70.135000

in3
            ltout       lcout       delta    
       min  214.215000  253.676000  39.461000
   typical  236.878000  280.513000  43.635000
       max  266.511000  315.606000  49.095000
```

mithro, 2018-07-15:
Does that help?

euripedesrocha, 2018-07-15:
Yes, that's what I was  looking for.

XVilka, 2018-12-03:
@euripedesrocha were there any updates on this?

euripedesrocha, 2018-12-03:
Sadly I had little time to work on this. I intend to pu some effort in january, If you want to lead the effort I'll be glad to provide you any help I can.

kgugala, 2019-04-05:
python SDF parser being developed in https://github.com/SymbiFlow/python-sdf-timing
SDF producer is a part of this PR https://github.com/SymbiFlow/prjxray/pull/706

kgugala, 2019-04-10:
The tool is more or less ready. I'm pretty sure it has bugs, and does not handle all the SDF spec. Will fix the bugs as they appear in testing.

mithro, 2019-04-10:
@kgugala I think we should change this bug to "Improve the python-sdf-timing library", some improvements that I can think of;
 - [ ] Add API documentation and publish on readthedocs
 - [ ] More extensive testing against specification
 - [ ] Output support
 - [ ] Example usage programs

mithro, 2019-04-10:
@rochus-keller - FYI There is a C++ library for parsing sdf files at https://github.com/kmurray/libsdfparse -- it might be worth adding SDF support to VerilogCreator if it isn't too much trouble.