f4pga/ideas #18: Create Python library for parsing / producing BLIF (and eBLIF) files

mithro, 2018-06-24:
# Create a Python library for parsing / producing BLIF (and eBLIF) files

BLIF is the preferred method of design entry for many designers. The Berkeley Logic Interchange Format (BLIF) is a simple file format for exchanging sequential logic between programs.

eBLIF is an extension of the BLIF format to add features needed for using BLIF as part of an FPGA PnR flow created by Clifford Wolf. The best format is the [Verilog to Routing docs here](https://docs.verilogtorouting.org/en/latest/vpr/file_formats/#extended-blif-eblif).

BLIF is used by Yosys, Arachne-pnr and Verilog to Routing.

There might be [a starting library here](https://github.com/whereyoufindthis/blifparse).

## Expected results

A Python library published on PyPi which makes it easy to read and write BLIF files.

# Knowledge Prerequisites

 * Python

nturley, 2018-07-15:
I can work on this. I'm a little busy at the moment so my progress would be steady but slow.

Do we have a corpus of BLIF files to test against?

mithro, 2018-07-18:
@nturley Only requirement is we need a pure Python library. Looks like antlr4 might support that? https://github.com/antlr/antlr4/blob/master/doc/python-target.md

kc8apf, 2018-11-05:
I started on a new [BLIF parser](https://github.com/gaffe-logic/tree-sitter-blif) using GitHub's recently released [Tree Sitter](https://github.com/tree-sitter/tree-sitter) incremental parser framework.

JohnDMcMaster, 2018-11-11:
Nice! Not sure where our BLIF roadmap is, but we'll take a look if we start poking around there