f4pga/ideas #19: Lowlevel LLVM-like language as HDL middle layer

XVilka, 2018-11-16:
Basic idea is to create something low-level, like LLVM Bitcode or WebAssembly, to create the HDL compilers emit the code in this format, which will be fed to the synthesizers after. This will allow to add 1) targets easier 2) HDL frontends easier.

XVilka, 2018-11-16:
Related https://github.com/SymbiFlow/prjxray/issues/58

mithro, 2018-11-16:
There is also https://bar.eecs.berkeley.edu/projects/firrtl.html
> FIRRTL (A Flexible Intermediate Representation for RTL) is motivated by the success of Chisel and has two parts: 1) a specification of the formalized elaborated graph that the Chisel DSL produces, prior to any simplification, and 2) a library of micro-passes that are used to simplify, transform, or specialize arbitrary FIRRTL graphs.

Nic30, 2018-11-16:
There is also [coreir](https://github.com/rdaly525/coreir).

The open hardware community really needs such a language, It is quite clear. I think that FIRRTL is good enough and already has large community. It would be great to have FIRRTL library for some languages like C++/Python/Java managed by creators of FIRRTL. If such a library exists for Python3 It would save me a lot of time and it will ensure the better compatibility in future.

I have some "hdl transpiler" [HWT](https://github.com/Nic30/hwt) (meta-HDL + verification env.) as you found and I will probably use FIRRTL as in backend in next release.
I plan to break  int small libraries and donate it to other projects like [magma](https://github.com/phanrahan/magma) or [SpinalHDL](https://github.com/SpinalHDL/SpinalHDL).

There are many hardware generators, for example [P4-to-VHDL](https://www.liberouter.org/wp-content/uploads/2016/05/p4.pdf). Problem is that the autors of such a tools have to implemented lots of same code over and over. The project are completely incompatible. It is not just FPGA-like assembly language and to VHDL/Verilog/SystemC conversions. There are graph algorithms which literary everyone has to implemented, like searching of isomorphic graphs etc.

@XVilka I think that there is a great opportunity in abstract specification of interface functions rather than in next FPGA-like assembly language. I have experimented with automatically generated DMA engines. The results are promising. The goal is to allow transaction level description with some extra meta-information specified by user. Not entirely the HLS bus some kind of metalanguage and automatic pipeline generator.

I think the first step is to create some consortium-like group for driving of the development of FIRRTL libraries in different programming languages and for management of libraries which are using FIRRTL. There is a very long road ahead (verifications, simulator-driver interfaces, analysis, HLS, architecture retargeting...), we need to act fast.

What ever you end up doing I am ready to help.

seldridge, 2018-11-16:
@Nic30: good points. 

- Python/C/C++/MyLanguage libraries for FIRRTL seems to make sense to improve the ease of frontend/backend development. The existing antlr and protobuf specifications for FIRRTL may already ease this.
- Yes, the idea is to not have everyone write the same dead code elimination/constant propagation passes... 

On the Chisel/FIRRTL side, we're trying to come up with some clear governance around Chisel/FIRRTL. We have Reviewer Guidelines upstreamed and a Constitution and Contributions PR'd. This is taking place on the [chisel-lang-governance](https://github.com/freechipsproject/chisel-lang-governance) repo and all of this is open to issues/PRs. The intent is to eventually have a catching organization/consortium for these projects. This is in its nascent form as the [Free Chips Project GitHub Organization](https://github.com/freechipsproject/). However, this primarily exists on GitHub at this time...

We just wrapped up the first Chisel Community Conference in the Bay Area and are planning to send out a form to identify what areas of development or non-development (testing upstream on big internal projects, outreach, etc.) individuals would be willing to help out on. This is expected to be posted on the [chisel-users](https://groups.google.com/forum/#!forum/chisel-users) and [chisel-announce](https://groups.google.com/forum/#!forum/chisel-announce) mailing lists, but we can publicize this more widely (i.e., here).

dvc94ch, 2018-11-17:
> @drom and I have started building out parsers for BLIF (
> https://github.com/gaffe-logic/tree-sitter-blif) and Verilog (
> https://github.com/drom/tree-sitter-verilog) based on
> http://github.com/tree-sitter/tree-sitter. So far, tree-sitter feels
> easier to develop a grammar and tests in. The result is a pure-C,
> incremental parser that already has bindings to Rust, Haskell, JavaScript,
> and Ruby. Higher-level libraries that convert the resulting ASTs into a
> language-specific, semantic representation will still need to be built
> after the parsers are ready.
>
> If someone wrote a tree-sitter parser for FIRRTL, we'd be well on our way
> toward having a reusable library that works both for toolchain components
> as well as editors and language servers.
>
> As for having a single organization, currently there are many: Free Chips
> Project, SymbiFlow, YosysHQ, Gaffe Logic, FOSSi. Most of these are loose
> affiliations related to a single tool or family of tools. I'm unclear on
> any benefit to having a single governing body for these widely-varied
> projects. Certainly having a focused place for community discussions would
> be helpful. Some of the larger projects might benefit from clearer
> governance. If someone wants to talk about different governance models,
> licensing, and foundations, please get in touch. I spent ~1yr working with
> Linux Foundation to get OpenBMC setup as an independent project with
> contributors from IBM, Intel, Facebook, Google, and Microsoft. I can
> definitely assist in getting everyone in touch with the major players in
> the free and open software foundations.
>
> Rick
>
> On Fri, Nov 16, 2018 at 1:11 PM Schuyler Eldridge <
> notifications@github.com>

Hmm it's been a while since I looked at the firrtl spec, but I seem to
recall that there is one implicit clock thereby not supporting multiple
clock domains and not making it very useful for FPGAs.

Nic30, 2018-11-17:
@kc8apf why are you not using ANTRL4 C++ runtime?

Nic30, 2018-11-17:
@dvc94ch  Is there really possible to have only one clock domain ?

dvc94ch, 2018-11-17:
Also if someone is writing a hdl in xyz, they may not like introducing a
dependency on the jvm. It should be able to link statically or dynamically
to a compiler toolkit imo, but maybe that's just me...

XVilka, 2018-11-17:
Agree, ideal implementation should be in C/C++, so every other
language/framework can link them very easily.

seldridge, 2018-11-17:
@ dvc94ch, @Nic30: FIRRTL supports arbitrary numbers of clocks and, consequently, you can use FIRRTL IR to write circuits with multiple clock domains.

You may have been thinking of the default behavior of Chisel modules where one implicit clock and one implicit reset is included. However, there's now a type hierarchy of modules, allowing you to easily remove the implicit clock and reset if you want/need more control. If you want more clocks, you add them as IO and can use scoping to control how they get used. The wiki page for Chisel3 goes into some examples that may help explain what that API looks like: https://github.com/freechipsproject/chisel3/wiki

kc8apf, 2018-11-17:
> Maybe we should split the topic of parsers and low level IR? While they
> are somewhat related I think there are definitely different goals here.

Sure. Parsers are a somewhat later concern. Any IR will need a detailed
spec to enable a variety of implementations. An existing grammar (ANTLR or
otherwise) is insufficient as a spec though I expect an IR to have a very
solid parser and generator for widespread adoption to occur.

dvc94ch, 2018-11-17:
@seldridge I'll have to try chisel, firrtl again sometime. Spinalhdl has added preliminary support for formal assertions etc. That would also be awesome to have support for in an ir.

Nic30, 2018-11-18:
@seldridge After I finish UVM like API for Verilator (+- 1 month) I can help with C++/Python FIRRTL. I have experience with ANTRL4, hardware design, etc. It would be great if anyone else can join as well.

Nic30, 2018-11-19:
Also there is a https://github.com/drom/awesome-hdl.
It would be great if anyone could just read about available tools on some well known place.
I think that now your are more or less lucky if you found some tool.
Also more options can better predict behavior of userbase in opensource projects.

XVilka, 2018-11-21:
It might worth to create a GitHub organization for this, so all different projects's maintainers will have an access to developing the common library/libraries.

kraigher, 2018-11-22:
FYI: I am working on an open source VHDL parser/language server written in Rust (https://github.com/kraigher/rust_hdl/). So If you need to parse VHDL code you are welcome to use/contribute to my project.

XVilka, 2018-11-22:
Who is familiar with FIRRTL? @christiaanb [asking](https://github.com/clash-lang/clash-compiler/issues/417#issuecomment-440974456) "It's very unclear whether FIRRTL has asynchronous resets, so I don't know whether it can be a target."

christiaanb, 2018-11-22:
As @XVilka mentions: I'm very interested as to whether the registers generated by FIRRTL have synchronous or asynchronous resets. [Clash](https://clash-lang.org) has support for both reset strategies; given that in FPGA land, on Intel/Altera fabric, registers have native asynchronous resets, while on Xilinx fabric, registers have native synchronous reset ports.

XVilka, 2018-11-22:
Apart from missing asynchronous resets there are also these issues in FIRRTL (by @Dolu1990 in https://github.com/SpinalHDL/SpinalHDL/issues/158#issuecomment-440996391 ):

- Quite weak clock domain support (no falling edge, no active low reset, no FPGA bistream reset)
- As far i know there is now support for Signal/components attributs (Quite often use by synthesis tools to give directives as KEEP, RAM_STYLE, ...)
- Inability to assign some bits of a combinatorial signal. For example `myWire(4 downto 2) := miaou`
- Driven by the Chisel needs (for the moment, which is a subset of what everybody need)

Dolu1990, 2018-11-22:
In short, this is a common issue with Chisel/FIRRTL :
freechipsproject/firrtl#219 => **low priority**

it realy annoys me.

seldridge, 2018-11-22:
#### Blackbox Escape Hatch

The robust blackbox support in Chisel and backed by FIRRTL `ExtModule` acts as an escape hatch for almost everything stemming from, "I can do this in Verilog, but not Chisel/FIRRTL." This handles a lot of either the weird exceptional cases or known feature requests like asynchonous resets that we're intending to support.

#### Asynchronous Resets

Yes, Chisel and FIRRTL do not _natively_ support asynchronous resets. There's tracking issues about this on the Chisel repository here:
  - https://github.com/freechipsproject/chisel3/issues/343
  - https://github.com/freechipsproject/firrtl/issues/219

These aren't terrifically hard to do, but it just hasn't gotten done because there's a good-enough workaround that Rocket Chip has used extensively via BlackBoxes. Their specific implementation is linked below:
  - [`AsyncResetReg` ](https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/util/AsyncResetReg.scala)

Currently FIRRTL treats Chisel-land resets as one-width `UInt`s, so they're always synchronous. There's an ongoing discussion on what to do with the FIRRTL IR type hierarchy to best support this. (Does Reset subclass Clock? Is it something else?)

The criticism of ["you can't use Chisel/FIRRTL in an ASIC chip"](https://github.com/freechipsproject/firrtl/issues/219#issuecomment-370652801) is mostly overblown with experimental evidence. SiFive, Berkeley, Google, (others who haven't announced publicly), have taped out a lot of chips and haven't seen this as a showstopper. There's been some inertia with asynchronous resets, though, as once some users have a good solution that works there's less pressure on the developers to implement the better solution. In short: it's an issue, we're working on it... if that timeframe doesn't work and you have ideas on how to improve things and want to help, we'd love to get you involved, though! (:wave: @Dolu1990)

#### SpinalHDL Criticisms

Spinal is (and correct me if I'm wrong @Dolu1990) a rewrite of Chisel2 implementing a lot of feature requests and fixing a lot of the issues that @Dolu1990 hit with Chisel2. If we can get some of these ideas  into Chisel3, we'd be happy to do it. However, there's a limited number of developers...

Responding inline to these:

> - Quite weak clock domain support (no falling edge, no active low reset, no FPGA bistream reset)

There's robust support for multiple clock domains ([chisel3/wiki/Multiple-Clock-Domains](https://github.com/freechipsproject/chisel3/wiki/Multiple-Clock-Domains)). @azidar has just PR'd [clock domain analysis support](https://github.com/freechipsproject/firrtl/pull/937). 

There's been a discussion on how to support neg-edge triggered things in:
- https://github.com/freechipsproject/firrtl/issues/695

This is on the agenda, but there hasn't been substantial need (i.e., developer pressure) outside of things which are going to be Blackboxed anyway (e.g., JTAG). It's also not clear how to best expose this to users via a good API so that they don't go nuts with it like they do in Verilog. 

> As far i know there is now support for Signal/components attributs (Quite often use by synthesis tools to give directives as KEEP, RAM_STYLE, ...)

There have been discussions about this. Basically, you'd like to be able to add custom emission to the Verilog emitter. There's an active issue here:
  - https://github.com/freechipsproject/firrtl/issues/941

> Inability to assign some bits of a combinatorial signal. For example myWire(4 downto 2) := miaou

Subword assignment is not supported, but there's a pattern to do this: 
  - [chisel3/wiki/Cookbook#how-do-i-do-subword-assignment](https://github.com/freechipsproject/chisel3/wiki/Cookbook#how-do-i-do-subword-assignment-assign-to-some-bits-in-a-uint)

> Driven by the Chisel needs (for the moment, which is a subset of what everybody need)

With limited developers, we've got to prioritize what developers are willing to drive. Consequently, if there's some feature you desperately need... we'd love to get you as a developer and have you drive it.

Summarily, we really, really want you as developers! The current developers are either grad students or engineers who can't work on Chisel/FIRRTL full-time. If there's a feature that you want, please file issues, throw PR's at us, or attend developer meetings (currently Fridays 1300--1500 PST/-0700, skipping this week due to US Thanksgiving). I can get you more info about the latter if you need.

Alternatively, if you, your team, or your company has a great idea on a better or alternative way to do it, go build it! If you'd like to provide inter-operability with FIRRTL (or Chisel in some way) that'd be awesome, though. :blush:

Dolu1990, 2018-11-22:
@seldridge 

**About BB and async reset :** 
Blackbox isn't a robust workaround for asyncronus reset, as with some technologies, the default is to use asyncronus reset.  (I don't see myself using blackbox everywhere in my code base)

Also using blackbox for elements like FF is quite enoying, as you have to manualy drive all the signals of the FF, i mean, you can't infer the FF behaviour in a friendly way via if statements.

Using BlackBox for FF also break in part the possibility to reuse FIRRTL optimisation passes.

It would actualy also be painefull to import at none FIRRTL netlist which use as default asyncreset into FIRRTL. Which is part of the subject of this issue.

I agree that black box is a usefull tool, but registers are too important for it.

**About Subword assignment**

I know there is workarounds, but the issue here, is if we want FIRRTL to be a universal LLVM of hardware, we probably need to support this kind of feature.

**General statement about hardware open-source tools**
I know that developping stuff require a lot of time, ressources are limited, and all that kind of things. But if we want to provide some quality tools to make everybody happy, if we realy want to make a viable alternative in that freezing hardware world, we can't stick to our personnal/projects needs, because that's just not good enough.

seldridge, 2018-11-22:
@Dolu1990: thanks for the feedback and thoughts!

- Yeah, __BB/Async__: this is the temporary solution. A more robust one is in the works. An RFC or discussion on the tracking issue about the API for this would be good.
- Maybe @azidar has some thoughts about __subword assignment__. It's difficult to think that subword assignment is a killer for an IR since it's a specification choice and the frontend needs to emit things to the spec. The mid-end/compiler doesn't care too much. I may be wrong, however. This is likely worth an RFC so that we can start evaluating options.
- Totally agree about __open-source tools__. On the Chisel/FIRRTL side we're trying to make it and the process as open as possible, attractive to users and developers, implement formal governance, etc. With the developers supporting it, we're not treating it as a personal project as we have users and companies (our own and others) depending on it. But, I (and the developers) have biases. If it's coming across as a personal project, tell us! :smile: If you have any feedback about this process, biases we're expressing, or ways that we can improve the developer interaction process, we'd be happy to hear it and try to address it.

Apologies for moving the discussion somewhat off-topic.

drom, 2018-11-22:
I think Verilog itself is the best HDL IR. It can be low level enough to become a netlist, or high level enough to be a human-readable-writable language. We just need to get standard AST representation of it and then develop lowering transformations when we can emit a valid Verilog in the middle.

Dolu1990, 2018-11-22:
@seldridge <3 ^.^

So about the subword assignement, it's all about making easy to export things into the FIRRTL format without too much transformation. Getting from a netlist with subword assignement to one without it require some quite specific transformations. It's not a IR killer, i agree.

About the community aspect, that's great, it's realy a good thing to open the developpement.

@JohnDMcMaster 
But the event driven paradigm add a lot of noise which has nothing to do with an digital logic hardware netlist, does it ?

I mean, if i look at FIRRTL and Verilog as a IR, that's the main thing which appear to me. So, that's maybe one thing to talk about.

dvc94ch, 2018-11-22:
The last time I tried doing some fpga design, I wasted a bunch of time trying to get migen to generate verilog that verilator would not complain about and then my changes where basically not accepted because every time you make a change 10 other verilog tools start chocking on it. At that point I think I may aswell just write verilog in a way that works for the tools I'm currently using. So all that using verilog as an ir means that we can just aswell give up on writing any other hdl's and just continue using verilog. The upside is we don't need an ir either so a lot of work saved.

seldridge, 2018-11-22:
@JohnDMcMaster: I've thought that Verilog as IR is tractable if you define a very restricted subset of Verilog as the IR. This has the plain difficulty of parsing the Verilog and dealing with all the complicated legacy Verilog semantics that live on due to it's history as a parallel event language. E.g., [Verific Design Automation](https://www.verific.com/) makes money _just selling Verilog/VHDL parser licenses_. An analogous approach would be to use a restricted subset of C/C++ as program IR. It's tractable, but likely difficult.

Relatedly, I've used [Verilog-Perl](https://www.veripool.org/wiki/verilog-perl) to write transforms on FIRRTL generated Verilog. This only worked because of the low-level, IR-like nature of the output Verilog. This approach breaks down if you try to get it to work on arbitrary input Verilog/SystemVerilog that exercises all language features.

I've then personally concluded that defining a clean circuit IR was a better approach.

(Unrelated note @JohnDMcMaster: I've always thought [WaveDrom](https://github.com/wavedrom/wavedrom) was super neat!)

@dvc94ch: Verilog design freedom would still exist, even in the presence of an IR. Excusing another software analogy: if you want to write an interrupt service routine using assembly because that gives you the control you need, that should work and continues to work. The utility of program IR comes when building compilers for high level languages targeting assembly. However, if you want to try to optimize assembly, inline assembly, or do interprocedural/link-time optimization, a frontend can read that assembly into program IR and work with it. Summarily, if Verilog works, use it! :+1:

dvc94ch, 2018-11-22:
Well I think I've given up on fpga's. There is plenty of fun software to write that doesn't have to do with fpga's. and my point was that a high level language has different semantics than the assembler. migen's semantics for example are intimately tied to verilog's.

Nic30, 2018-11-22:
> I think Verilog itself is the best HDL IR. 

@drom [hdlConvertor](https://github.com/Nic30/hdlConvertor) I tried to. It has no use as intermediate language because raw Verilog does not fit well into graph database. Also Verilog has many it's own problems. f.e. ROM has to be written as switch statement in old versions etc.

But you have the point. We do not need to have new language, we need unified API to represent HDL, parse it and perform manipulations over it.

I would like to use some sort of serializable in-memory graph database. But there is not any suitable known to me. (Things like Neo4j are too slow and not optimized for HDL)

Also parsing of VHDL is incredibly slow, Verilog is faster, but still... 

@seldridge If we cant have an async resets/latches/falling edge regs, can we at least add black-boxes into FIRRTL library? (It seems to me that there is a problem. You can not specify the data-width for such a blacbox.)
Because otherwise FIRRTL can not be used for development of IO and it's usability for Intel FPGAs is limited. Also this seems to me like a best compromise for everyone which does not break anything. I think that sooner or later you would be forced to add it anyway. F.e. ARM SoCs have many async. transitions. And it is only question of time when you really will have to write this in Chisel and you will have to use latches.

As you say it is not hard to create your black box and import it but how can automatic analysis work when everyone has many black box entities in own design? How are things compatible then? Will the HDL be copied every time  the Verilog is generated? Can there be a name collision? Is name colision checked when FIRRTL -> Verilog collision?

@dvc94ch We are not doing it because it is easy. There are project which can choose ASIC or FPGA because other option does not exists. And question now is how do we seed up HW development. 

> migen's semantics for example are intimately tied to verilog's

I also see this a problem. But they are satisfied.

kc8apf, 2018-11-22:
My desire for an IR is to have something that is fairly simple and
straightforward to translate to a variety of FPGA architectures and ASIC
cells. Verilog and VHDL are much higher-level and would be implemented in
terms of the IR similar to how C and Rust are implemented in terms of LLVM
IR. BLIF is much closer to what I'm imagining.

I can see a desire for an even higher-level IR that offers additional,
common constructs. HDLs might target that high-level IR and a common set of
transforms translate it to the low-level IR. That way every HDL frontend
doesn't have to reinvent an adder and backends don't need to recreate how
to construct one out of LUTs. Another way to think about this is a standard
library for the low-level IR.

seldridge, 2018-11-22:
> @seldridge If we cant have an async resets/latches/falling edge regs, can we at least add black-boxes into FIRRTL library? (It seems to me that there is a problem. You can not specify the data-width for such a blacbox.)

To clarify, you _can_ have async resets, latches, falling edge registers... just not yet. These are planned to be implemented, just these are feature requests at this point.

FIRRTL does have blackboxes via the `ExtModule` IR construct. This allows you to instantiate a module and not define it. This can be filled in by a Verilog BlackBox (possibly parameterized) or via inline compilation. E.g., you can use Chisel (or Magma, etc.) to generate FIRRTL/Verilog or define what this is.

> Because otherwise FIRRTL can not be used for development of IO and it's usability for Intel FPGAs is limited. Also this seems to me like a best compromise for everyone which does not break anything. [...]

Latches as an _explicit_ construct have been discussed, just not implemented. The question here is the API.

> As you say it is not hard to create your black box and import it but how can automatic analysis work when everyone has many black box entities in own design? How are things compatible then? Will the [...]

Lots of questions. Analysis with blackboxes is difficult because the analysis can't look inside into an `ExtModule`. Granted, most of what you need for digital design is available. The intent is to fill out both the Chisel-level API and the backing FIRRTL support so that you have a full toolbox to build whatever you need. Blackboxes fill that hole while things are being implemented.

`ExtModule`s would only be defined once, not multiple times.

There's extensive support in FIRRTL for avoiding name collisions during lowering (flattening aggregates) as well as for generating temporaries, etc. There's internal APIs for making it easy for transform writers to get it right.

FIRRTL collisions with Verilog keywords is taken care of, e.g., a FIRRTL module named `always` (I wrote that transform). FIRRTL name collisions with `ExtModule`s are taken care of by looking at the port names (which is enough to clear name collisions). Collisions during module inlining or flattening are also taken care of (I took care of a refactor that makes sure this works correctly).

rochus-keller, 2018-11-23:
> I think Verilog itself is the best HDL IR. 
Agree. But then I would highly recommend to restrict it to Verilog 2005 or 2001 (or even a defined subset of IEEE 1364), with no SystemVerilog protuberances.

Nic30, 2018-11-23:
@rochus-keller

> EDIF 

I know some tools can use it, but there is a problem, there have to be explicit instances for example for adder. The result is more or less target technology locked. I do not think that we can maintain support for all FPGAs and ASIC manufacturing processes.

EDIF readability is a joke. It does not support more files and is quite large.
I mean EDIF is way to hell blessed by Xilinx and others back in 2005 if I remember correctly.

rochus-keller, 2018-11-23:
> EDIF - The result is more or less target technology locked. I do not think that we can maintain support for all FPGAs and ASIC manufacturing processes.

You seem to mix up the format with some unfortunately bad implementations. I actually mentioned EDIF primarily because BLIF was already mentioned in the discussion and both have overlapping concerns and can be converted to each other.

Nic30, 2018-11-24:
Ok, it seems to me that many people want to use Verilog from various reason. Can someone create specification which features of Verilog can be used to make it ~equivalent with FIRRTL so potential conversion can be performed?

Also how to import the peoples which are working on HDL generators that they should use this?

Dolu1990, 2018-11-24:
The issue i have with Verilog as a IR that you can work on is : 
- If you want to work on a IR which keep much of the use sementic (mainly if statement/switch case stuff) you have to deal with this stupid event driven paradigm which split everything everywhere with always blocks.
- If you use verilog as a lower level IR, you lose much of the possibilities to have "abstract" transformation phases. And to export another IR to this low level verilog IR you need to lower the source  IR, which is quite borring and technology specific

Nic30, 2018-11-24:
@Dolu1990

Maybe IR should be really low level language and the "abstract" transformation phases should be part of the library.

Also it is required what "abstract" transformation phases are.
For example this transformations is possible to implement in library for generating of IR and it is not required to have support for this in IR.

* blocking code -> FSM in HW
* datapath -> pipeline  (to be usefull requires info about target plat.)
* regular statements -> processes with generated sensitivity list in HW

Would you add any other?

rochus-keller, 2018-11-24:
I had a quick look on Google and found this link: http://www.iuma.ulpgc.es/~nunez/clases-FdC/verilog/Verilog-IEEE-1364.pdf. Maybe you want to have a look at it as a starting point. You find the syntax diff to VL 01 on page 29. Probably it's possible to kick out even more things and to make use of attributes for higher-level information. A cross-correlation with FIRRTL seems to be quite some effort.

rochus-keller, 2018-11-24:
Have a look at it first. It's not the full Verilog standard, but a defined synthesizable subset. From my point of view it's a good starting point for an even more stripped-down version.

rochus-keller, 2018-11-24:
Yet another good reason to stay with Verilog and not to switch to the academic flavour of the month.

Nic30, 2018-11-24:
@rochus-keller RISC-V and Chisel3 (including FIRRTL) seems to you as academic flavour of the month?

@dvc94ch Thank you for the information, I was thinking that Yosys also use only subset of Verilog. I will look into it.

Dolu1990, 2018-11-24:
When i look my calendar, i can see that we are nearly in 2019, and we are still facing VHDL/Verilog as glorified industry standard. This is depressing XD

> lavour of the month.

@rochus-keller That's overstated XD

Nic30, 2018-11-24:
Also @rochus-keller you are kind of right. My question how to provide universal API to HDL and persecute others to use it in the favour of theyr's partially working solution. Using Yosys seems to be reasonable step.

rochus-keller, 2018-11-24:
> seems to you as academic flavour of the month?

Oh, you're right, FIRRTL is of course the flavour of the penultimate month - released sept. 18  ;-) 
RISC-V is a good thing and well adopted in industry, but I'm not aware that Chisel is really used somewhere outside academia; maybe you have references who give other evidence.

> i can see that we are nearly in 2019, and we are still facing VHDL/Verilog as glorified industry standard.

Well, the same then applies to C, Unix, OO, ML and some other cool stuff.

Dolu1990, 2018-11-24:
> Well, the same then applies to C, Unix, OO, ML and some other cool stuff.
The difference is that C was designed as a programming language, while VHDL/Verilog wheren't designed as languages to describe RTL. Even when they are used to write testbench, they aren't that good.

> Oh, you're right, FIRRTL is of course the flavour of the penultimate month - released sept. 18 ;-)
> RISC-V is a good thing and well adopted in industry, but I'm not aware that Chisel is really used somewhere outside academia; maybe you have references who give other evidence.
Maybe part of the issue is in Chisel, but in all cases the industry itself has a big issue, VHDL/Verilog are real bottlenecks, but it is quite hard to understand/fell those bottlenecks unless you deeply experiment a different ecosystem. Also, most hardware enginner don't have the training to understand the advanced capabilities of Chisel, and Chisel itself isn't advertising too much on advanced capabilities to not afraid people with software generators. Result of that is that many people think Chisel is just another Verilog but in scala, which is fondamentaly wrong.

seldridge, 2018-11-24:
@rochus-keller: Chisel/FIRRTL users that I'm aware of, primarily as I sat through some talks just recently:
  - SiFive (reference likely not needed)
  - Intel [[1](https://youtu.be/IZeUHzukStE)], [[2](https://youtu.be/FktjrjRVBoY)], [[3](https://youtu.be/FktjrjRVBoY)]
  - Google [[4](https://youtu.be/x85342Cny8c)]
  - IBM (I'm using it within IBM Research, but here's one usage [[5](https://github.com/IBM/chiffre/)])

Full product teams adopting it is relegated to SiFive and the Google Edge TPU product.

For some historical background: Chisel and FIRRTL grew out of DARPA's (ongoing) CRAFT program. Chisel has been around since 2012. FIRRTL grew out of a rewrite of Chisel 2 to add an IR, creating the Chisl front-end. FIRRTL has been a project since, as far as I'm aware ~2014 (but, possibly earlier).

dvc94ch, 2018-11-24:
> Well, the same then applies to C, Unix, OO, ML and some other cool stuff.

I don't understand what you mean. I don't know why ML is an industry standard. Assuming you mean the language and not Machine Learning it's popular with compiler writers. Which I assume you think is a waste of time because we already have C, so why build a new compiler. I'm sure that if Ritchie had that mentality we might still be writting assembler or even better punching holes into a card.

rochus-keller, 2018-11-24:
Thanks for the vivid discussions. I look forward reading the provided information. But it's getting off topic. You asked for opinions concerning "Lowlevel LLVM-like language as HDL middle layer". I put my arguments on record. I think we can agree that Chisel is no "HDL middle layer" candidate, regardless where it is used, and I would - from my current understanding - also FIRRTL rather see as an alternative offer to Verilog/SystemVerilog/VHDL, not as a "lowlevel HDL .. like LLVM Bitcode or WebAssembly"; on that level you rather find BLIF, EDIF or a primitive Verilog subset; but don't let me keep you from exploring new grounds ;-)

Dolu1990, 2018-11-24:
@rochus-keller FIRRTL isn't an alternative to VHDL/Verilog/SystemVerilog. It was designed to represent a machine generated netlist. All the elaboration feature like functions, loop, parameters are gone from it.
From memory, i remember that there is 3 level of feature defined in FIRRTL, probably @seldridge can expose them better than me, it would probably add some interresting axes of thinking.

rochus-keller, 2018-11-24:
> It was designed to represent a machine generated netlist.

To cite from Li, Izraelevitz, Bachrach (2018): Specification for the FIRRTL Language, Version 0.2, section 1.2 Design Philosophy, page 6: _FIRRTL represents the standardized elaborated circuit that the Chisel HDL produces. FIRRTL represents the circuit immediately after Chisel’s elaboration but before any circuit simplification. It is designed to resemble the Chisel HDL after all meta-programming has executed._
But I think we can leave this topic for now.

dvc94ch, 2018-11-24:
With [electron-lang.org](https://electron-lang.org) I thought that if I stick to a purely structural description, it is also possible to describe mixed signal circuits using the same ir. electron-lang is pretty much dead at the moment (mostly because nice stuff in my head often turns out into an imperfect reality) and didn't reach all of it's goals. With some contortions it generates yosys compatible json and uses yosys to convert it to verilog.

Nic30, 2018-11-26:
About the needs of some projects I have seen (non authoritative response, fix me if I am wrong):

[Chisel3](https://github.com/freechipsproject/chisel3/wiki)
  * netlist of statements (process without explicit sensitivity)?
[HWT](https://github.com/Nic30/hwt)
  * netlist of statements
[magma](https://github.com/phanrahan/magma)
  * netlist of statements
[MyHDL](https://github.com/myhdl/myhdl)
  * processes 
[SpinalHDL](https://github.com/SpinalHDL/SpinalHDL)
  * netlist of statements
[Veriloggen](https://github.com/PyHDI/veriloggen)
  *  whole Verilog
[Yosys](https://github.com/YosysHQ/yosys)
  * netlist of hw processes ?

XVilka, 2018-11-26:
I created an organization and invited you as an owners @Nic30 @seldridge , so you can do whatever you like with access/whatever. I also created https://github.com/HardwareIR/meta for this discussion, so probably some summary can formed.

XVilka, 2018-11-27:
@Nic30 updated the top post with information from your comments, so it will be visible from the start. Too bad it is impossible to allow just edit it.

XVilka, 2018-11-27:
Was also pointed to http://rsim.cs.illinois.edu/Pubs/17-PPOPP-HPVM.pdf - worth checking out.

benreynwar, 2018-11-29:
I'm in favour of going with either:
1) FIRTTL simply because there's already a bunch of work done, and it doesn't seem like there are any issues that can't be fixed with a bit of work.
2) A very restricted subset of verilog.

I think a common IR will also be useful in taking timing reports, error messages and other tool outputs and transforming them up to the level of the HDL.  At the moment it's difficult to do the timing analysis and the verification without going down to the verilog level.  This negates a lot of the benefits of the HDL since the generated verilog is more difficult to deal with.

Design
HDL -> IR -> verilog

Timing Reports, Error messages ....
verilog -> IR -> HDL

seldridge, 2018-11-29:
@benreynwar: It's hidden in a link above, but Intel has had some success doing IR-level analysis for area and timing: 
  - [intel/rapid-design-methods-for-developing-hardware-accelerators](https://github.com/intel/rapid-design-methods-for-developing-hardware-accelerators)
  - https://www.youtube.com/watch?v=FktjrjRVBoY (recent presentation video)

This isn't intended as a replacement for tape-in analysis. However, it's apparently proved useful for finding bad design decisions early, e.g., big combinational logic chains. I'm not sure how far you can push this idea before having to do synthesis/place-and-route, but it would seem that doing IR-level analysis would let designers iterate faster, produce better designs, and reduce a reliance on proprietary tools early in the design process.

Nic30, 2018-11-30:
I will collect results of our discussion in to some document after 12.8, (currently I am out of time).
So do not be shy to share any interesting projects or facts.

Also description of unique properties which makes some small project better than everything else is welcomed.
Also I do not know if there is some DSP library for chisel3 which actually works and does not requires to manually instantiate DSP blocks (like DSP48E1).

dvc94ch, 2018-11-30:
My non-expert opinion is the following:

There are two large application areas for FPGAs.

1. Specialized SoCs for embedded systems
2. Application specific hardware accelerators for offloading

For 1. a source language like chisel is appropriate.

For 2. maybe something like a standard library written in chisel and a
source language that translates to spirv (glsl) might be more appropriate.

Then there is the issue of tooling and what abstraction level that is most
suitable for a specific task.

I think that one ir won't fit all problems. I could imagine a set of three
ir's to work out.

1. Something close to spirv (subset/superset)
2. Something close to firrtl (superset)
3. A structural netlist ir similar to yosys json format (superset that also
has non digital circuits in mind)

dvc94ch, 2018-11-30:
FYI I have used antlr before and porting it to a new language is a non
trivial task that involves rewriting the runtime. After the discussion here
about tree-sitter, I gave it a try for parsing a small dsl for specifying
keybindings:

```
mode "command" {
  bind i -> mode("insert")
  bind p -> action("paste")
  bind SPC 2 -> action("focus-2")
}
```

And it's pretty neat. It's also very easy to add support for new languages,
since it has a very small C API surface.

On Fri, Nov 30, 2018, 13:31 David Craven <david@craven.ch wrote:

> My non-expert opinion is the following:
>
> There are two large application areas for FPGAs.
>
> 1. Specialized SoCs for embedded systems
> 2. Application specific hardware accelerators for offloading
>
> For [...]

[32 more replies]