f4pga/ideas #20: Verilog AST format

drom, 2018-11-26:
# Brief explanation

Open and Common format of Abstract Syntax Tree of Verilog program.

## Expected results

Tools ecosystem using the format.

# Detailed Explanation

Open HDL ecosystem needs more tools and libraries to deal with Verilog.
Common Abstract Syntax Tree data format will enable interoperabity.

Good collection of AST formats for different langages: https://astexplorer.net/

Set of core libraries:
  * Verilog parser --> AST
  * Verilog code generator <-- AST
  * AST traversal, query and utility libraries

Good example of ecosystem enabled by AST format is ESTools:
  * Parser : http://esprima.org
  * Linter : https://eslint.org
  * Compiler : https://babeljs.io
  * Code coverage : https://istanbul.js.org
  * Tool belt : https://github.com/estools

## Further reading

http://modularity.info/conference/2013/escodegen.html

THofstee, 2018-12-03:
I'm not necessarily advocating for this but https://github.com/PyHDI/Pyverilog and https://github.com/PyHDI/veriloggen exist. They don't support things I find useful like multidimensional arrays (all packed/unpacked dimensions need to be flattened to a single packed and a single unpacked), or array assignments if I remember correctly. Nor any SystemVerilog features. Also the documentation is, well, lacking.

rochus-keller, 2018-12-24:
In case you're interested: here is a complete Verilog 2005 parser https://github.com/rochus-keller/Verilog and here is a meta parser generator https://github.com/rochus-keller/VerilogEbnf which helps to not pollute the LL(1) syntax with implementation details. All written in C++. The current incarnation of the parser/AST is optimized for my VerilogCreator IDE.

drom, 2018-12-24:
I am working on Verilog 2017 parser here https://github.com/tree-sitter/tree-sitter-verilog

drom, 2018-12-25:
@rochus-keller I know it is hard.
The reason I am doing new parser grammar is that I want to use it with tree-sitter : https://tree-sitter.github.io/tree-sitter/ developed by GitHub for Atom IDE and code analysis.
It is WIP. about 20% of grammar.
@rochus-keller I would appreciate any help, guidance or feedback.

rochus-keller, 2018-12-25:
Do you really need SystemVerilog? From what I've seen even the big tools don't support every aspect let alone the most recent standard. The open source tools only implement a little subset, mostly asserts, which is what I will do next too.

drom, 2018-12-27:
Yes, I want SystemVerilog support at least at the parser level. It definitely will be implemented in stages.

rochus-keller, 2019-01-06:
I will extend VerilogEbnf by generators for PEG and GLR grammars, generate parsers for the original ambiguous grammar (with PackCC and Elkhound) and then make performance tests to be able to compare time and memory consumption with my existing LL(1) version of the parser, which only needs a few seconds for my ~6000 verilog test files. PEG has O(n2) and GLR O(n3) worst case time complexity but can cope with the ambiguous grammar - at least in principle. My current parser runs in linear time but it was a lot of work to modify the grammar (doing such modifications on the full SystemVerilog grammar looks like a life's work).