f4pga/ideas #21: Verilog IDE based on Qt Creator integrated with SymbiFlow

rochus-keller, 2018-12-22:
# [{{reference.repo}} #{{reference.number}}] {{title}}

More technical details at [Link to bug in the ](http://github.com/SymbiFlow/{{reference.repo}}/issues/{{reference.number}})

# Brief explanation

A short description of what the thing you want to do.

## Expected results

What the expected outcome from the project should be.

# Detailed Explanation

A much longer description of what the thing you want to do.

## Further reading

 * [Link to a PDF document](http://abc/abc.pdf)
 * [Link to a some HTML page](http://abc/abc.html)

# Knowledge Prerequisites

 * List of
 * what knowledge is required
 * to complete this project.

mithro, 2019-03-05:
Hi @rochus-keller,

It looks like you are already working on this project, so I'm going to close this issue.

Thanks for the reference,

Tim '@mithro' Ansell

closed 2019-03-05 by mithro

rochus-keller, 2019-03-05:
Hi mithro, 
yes I made progress; the IDE works and is being used; the most recent version also supports parts of SystemVerilog assertions (zipcpu and picorv32 also work in FORMAL mode). 
I don't mind if you close the issue; what do you think of the idea to use the IDE as a frontend for symbiflow?
Best
R.

mithro, 2019-03-07:
@rochus-keller At the moment we don't plan to include a frontend in SymbiFlow. We expect that people will use their own preferred editor.