f4pga/ideas #22: Participate to Google Summer of Code

XVilka, 2018-12-24:
Might be beneficial for the project, attract more attention and contributors.

The application starts in three weeks:
- January 15 20:00 UTC | Mentoring organizations can begin submitting applications to Google
- February 6 20:00 UTC | Mentoring organization application deadline

See https://developers.google.com/open-source/gsoc/timeline

programmerjake, 2019-01-26:
reminder that the deadline is quite soon

mithro, 2019-02-04:
I have started putting together an application for GSoC. The best thing people can do to help is to improve (and add) ideas in this issue tracker.

See some tips at https://github.com/sympy/sympy/wiki/GSoC-2018-Ideas and https://google.github.io/gsocguides/mentor/defining-a-project-ideas-list

mithro, 2019-02-04:
@kgugala @mgielda @pgielda - You are probably interested in this.

XVilka, 2019-02-05:
Deadline is tomorrow by the way.

mithro, 2019-02-05:
Just FYI - Deadline for applications is actually February 7, 2019 at 07:00 (Australian Eastern Daylight Time).

I believe we have put in an application. If we are not successful in our application we will look at participating via an organisation like [TimVideos](https://code.timvideos.us/home/) or [FOSSi Foundation](https://fossi-foundation.org/) (assuming they are accepted).

mithro, 2019-02-05:
(Wednesday, February 6th at 20:00 UTC for those not in my timezone :-)

XVilka, 2019-02-27:
Congratulations! You got selected 🥇  Closing then as solved.

https://summerofcode.withgoogle.com/organizations/4517422304854016/

closed 2019-02-27 by XVilka

XVilka, 2020-01-02:
To everyone involved - just a reminder that GSoC 2020 organization applications will be open on January 15 until February 6 - https://summerofcode.withgoogle.com/how-it-works/#timeline