f4pga/ideas #28: Get LiteX/VexRiscv supported in upstream Zephyr and update RISC-V Getting Started Guide

mgielda, 2019-04-23:
See https://github.com/zephyrproject-rtos/zephyr/pull/14915

and 

https://github.com/riscv/risc-v-getting-started-guide

This allows SymbiFlow to be used in the GSG.

mgielda, 2019-04-23:
This also includes LiteUART.

LiteEth will be done separately as there is still some work to do there.

PiotrZierhoffer, 2019-05-20:
This is already done.

Zephyr support for VexRiscv is merged here: https://github.com/zephyrproject-rtos/zephyr/commit/f9efca4b4f3ee81a1e36646ab52a020c5e2d14da

The Getting Started Guide is updated to point to mainline Renode and mainline Zephyr here: https://github.com/riscv/risc-v-getting-started-guide/commit/2f92d10cacebdaa9b762da2fb79fa7c46e71aa05

closed 2019-05-20 by mgielda