f4pga/ideas #33: Getting started with embedded RISC-V tutorial

mgielda, 2019-04-23:
# Tutorial for SymbiFlow with embedded RISC-V

# Brief explanation

Write a tutorial for getting started with SymbiFlow on an embedded RISC-V CPU on a small FPGA with Zephyr.

## Expected results

An easily reproducible tutorial which enables people to work with SymbiFlow on a relatively simple yet useful example system.

# Detailed Explanation

SymbiFlow can already work with Zephyr-enabled SoCs like LiteX/VexRiscv. Targeting Lattice iCE40, write a tutorial that lets people compile Zephyr, a bitstream that can run it, and then program an FPGA to see it in practice. As a second track, enable the same software to be run in Renode without modification if someone does not have an FPGA yet wants to work on the software side of things.

# Knowledge Prerequisites

* Sphinx
* bash

AlaaHussein501, 2020-03-15:
Hi, I found "Getting started with embedded RISC-V tutorial" project so interesting I want to participate in it. Although I'm at a beginner level, I'm willing to learn. Kindly provide me with any learning resources needed.

sordfish, 2020-05-27:
This is literally the use case I was looking for. 

I'm a complete FPGA, Riscv and Zephyr noob and don't have any hardware but I've got a project that I think this setup can go far with.

I'll be digging through what I can find about Renode and Zephyr and I'll make sure to makes notes.