f4pga/ideas #39: Support for cheapest FPGAs on the market

Harvie, 2020-01-28:
Hello,
i've searched the Aliexpress for absolute cheapest FPGA boards and i've found very cheap devices that i think might be worth supporting:

- Noname chinese bargain FPGAs
  - https://hackaday.com/2019/11/03/the-5-fpga/
  - gowin GW1N-1  (>1k LUT4, development board for $9, free worldwide shipping)
  - Anlogic EG4S20 (20k LUT4/LUT5 mix, development board for $18, free worldwide shipping)
- Cheap FPGA development boards
  - Intel/Altera Cyclone II EP2C5 (4.6k LUT, development board for $12, free worldwide shipping)
  - Intel/Altera Max II EPM240 (~200 Macrocell **CPLD**, development board for $7, free worldwide shipping)
- Bit more expensive, but still reachable aliexpress deals:
  - Intel/Altera Cyclone IV EP4CE10 (10k LUT, $24)
  - Intel/Altera Cyclone IV EP4CE6 (6k LUT, $19)
  - Xilinx Spartan6  XC6SLX16-FTG256 (14.4k LUT, $18)

I think that support for these inexpensive devices might help get FPGA to hobbyists and schools (including developing countries).

To be clear i don't mentioned ICEBreaker with Lattice ICE40 shipped for $30 USD from Aliexpress, which is my top favourite right now. But it is already supported by icestorm project, so no need to discuss that in this issue.

mithro, 2020-01-28:
@Harvie - You should check out the Colorlight 5A-75B which is an ECP5 based board available for O($15 USD) shipped from Ali Express. People are already working on documenting the board at https://github.com/q3k/chubby75/tree/master/5a-75b

mithro, 2020-01-28:
@pepijndevos was also working on the Gowin FPGAs at https://github.com/pepijndevos/apicula

mithro, 2020-01-28:
Oh, my Fomu board (https://j.mp/fomu-cs) is also sub $20 USD when bought in 100 unit quantity. We have an excellent workshop for it at https://workshop.fomu.im

Harvie, 2020-01-28:
@mithro :-) I am aware of TOMU/FOMU, saw some of your videos. Cool hardware! If i ever need project that fits to USB socket, i will surely use it.

Project Apicula sounds promissing. I'll surely take a look at it.

Haven't seen DiColorlight 5A-75B, but it looks funny :-D But that's obviously caused by the fact it's not actual development board, but rather LED matrix driver board, which can be repurposed for development needs...

Thanks for interresting info.

Harvie, 2020-01-28:
There is also some project looking at anlogic FPGAs: https://github.com/mmicko/prjtang

rjordans, 2020-03-13:
The Lattice MachXO series would be nice to see as well on the list.  The breakout board from Lattice is similar in price to the ICEBreaker but the flash based FPGA is also relatively easy to use in custom designs as it only really requires a power supply.

pepijndevos, 2020-03-14:
Yea for Gowin, there is a working proof of concept. There is a lot of work left to do. In terms of documenting, global clock nets and more primitives. In terms of software, a proper place and route target in vtr and/or nextpnr. (Currently using nextpnr-generic) I'm currently doing my master thesis so progress is extremely slow. I'm working on documenting timing.

YusefIiqbal, 2020-03-15:
Hello @Harvie , My name is Mohd. Yuse fIqbal, and I am a Third-Year student of Bachelor in Technology in Electronics and Communication Engineering from Jamia Millia Islamia. In my introductory coursework, I have developed an interest in Semiconductor modelling and also in Digital Circuit Design languages like Verilog, VHDL, RTL etc. If possible, I would love to start working under you on the project on this project "Support for cheapest FPGAs on the market"
Would it be possible to further discuss the project and my possible involvement in the project? And where to contribute and also, I need your help in creating my Gsoc Proposal and SOP. I look forward to hearing from you.
I look forward to hearing from you.

muhammadsalmanafzal, 2020-03-18:
Hi @Harvie , I'd also like a description about what this project is about and how can I contribute to it by taking part in GSoC???

mithro, 2020-03-18:
There are details at https://symbiflow.github.io/summer-of-code

Harvie, 2021-03-11:
I've just updated the list in the original post to highlight FPGAs available through JLCPCB SMD assembly.
Complete list at https://jlcpcb.com/parts/componentSearch?searchTxt=FPGA