f4pga/ideas #47: Updated the repositories to compile with the current copyright best practices

mithro, 2020-04-13:
* https://opensource.google/docs/releasing/preparing/#license-headers
 * https://opensource.google/docs/releasing/authors/

Specific things that need to be done;
 - [ ] `COPYING` file renamed to `LICENSE` file
 - [ ] The `LICENSE` file should start with - `Copyright 20XX The [Project Name] Authors.` -- See https://opensource.google/docs/releasing/authors/
    - [ ] Project X-Ray --> `The Project X-Ray Authors`
    - [ ] Project U-Ray --> `The Project U-Ray Authors`
    - [ ] Project Trellis --> `The Project Trellis Authors`
 - [ ] Add or update the `AUTHORS` files to have the text as shown at https://opensource.google/docs/releasing/authors/
 - [ ] Remove the un-needed `All rights reserved.` text.
 - [ ] Make sure all code files have the license specifier at the top of the file as shown at https://opensource.google/docs/releasing/licenses/
 - [ ] Add CI which checks for all this.

What other things are missing?

mithro, 2020-04-13:
Tracking a bunch of the status at https://github.com/orgs/SymbiFlow/projects/18

mithro, 2020-04-13:
FYI - @kgugala, @litghost, @acomodi, @mkurc-ant

daniellimws, 2020-04-26:
The CI scripts can be copied from https://github.com/SymbiFlow/fpga-tool-perf/tree/f8d143b73f670eea5efdbe96f1236d82998c896e/.github/travis

mithro, 2020-04-26:
@daniellimws - I wonder if we need something like a https://github.com/SymbiFlow/symbiflow-travis-common repo that contains scripts like the license check?