f4pga/ideas #50: Add an .editorconfig file to all SymbiFlow repositories

mithro, 2020-04-30:
[`.editorconfig` is a file](https://editorconfig.org/) which tells many editors how to set up their auto formatting configurations. See this example in the Yosys repository -> https://github.com/YosysHQ/yosys/blob/master/.editorconfig

We should create an `.editorconfig` for SymbiFlow projects and then add it to all our repositories.

We probably need a configurations for;
 - [ ] Python files
 - [ ] C++ files
 - [ ] TCL scripts
 - [ ] Bash scripts
 - [ ] Makefiles
 - [ ] cmake files
 - [ ] Verilog files
 - [ ] Exclude `third_party` directory

For Python files, it should match what `.yapf` settings we are using.
For C++ files it should match the Google C++ Style Guide.
For Verilog files it should match the LowRISC Verilog Style Guide.

What else have I missed?

mithro, 2020-04-30:
```
; EditorConfig is awesome: http://EditorConfig.org
; At this time of writing, EditorConfig plugins are currently available for:
;
;     Code::Blocks
;     Emacs
;     Geany
;     Gedit
;     jEdit
;     Notepad++
;     Sublime Text 2
;     TextMate
;     Vim
;     Visual Studio
```

FFY00, 2020-04-30:
editorconfig is very limited when it comes to language support, it mostly just allows you to configure the identation and some minor styling preferences.

I think it should be used unless there are better language specific alternatives. Python has flake8, C/C++ has clang-format, I am not sure about the other languages.

mithro, 2020-04-30:
@FFY00 - This is more about providing the editor with hints on things like indenting that are *consistent* with what the formatting tools do. We don't want a person's editor and the code formatting tools fighting.

FFY00, 2020-04-30:
Well, flake8 is not really the same thing, but clang-format can be used as a C/C++ specific editorconfig, but yes, it can also be uses as a formater. Anyway, as long as .editorconfig and the language specific tools are sync there is no issue in providing it, it is still a good addition :)