f4pga/ideas #51: Create a "common configuration" git repository which is auto-merged into SymbiFlow projects

mithro, 2020-04-30:
We are increasingly getting files that should be consistent across all SymbiFlow repositories.

We should create a "common configuration" git repository which is auto-merged into the various SymbiFlow projects. It should probably include;

 - [ ] Stuff around style formatting including;
     - [ ] `.editorconfig` file
     - [ ] Copyright / license checking tools (plus Travis CI support for checking them)
     - [ ] Auto-formatting tools - `clang-format`, `yapf`, etc (plus Travis CI support for checking them)
 - [ ] Documentation around policies
     - [ ] Code of conduct
     - [ ] Contribution guide
     - [ ] Issue templates
     - [ ] Common replies to things like missing DCO and stuff
     - [ ] etc
 - [ ] Documentation building and publishing scripts
     - [ ] The Sphinx documentation generation
     - [ ] ReadTheDocs publishing
 - [ ] Useful infrastructure / CI scripts
     - [ ] Tools for handling the complexity of git submodules
     - [ ] Tools for getting and setting up environments using conda
     - [ ] etc

This could be related to https://github.com/SymbiFlow/.github or potentially replace it.

mithro, 2020-04-30:
FYI - @daniellimws - You have been doing a lot of stuff around the documentation side of things, any thoughts on how to make that work?

mithro, 2020-04-30:
This should probably use [`git subtrees`](https://blog.developer.atlassian.com/the-power-of-git-subtree/) and there be a bot to auto merge this into our repositories.

mithro, 2020-04-30:
#50 is related.

mithro, 2020-04-30:
https://github.com/SymbiFlow/ideas/issues/49 is also related.

daniellimws, 2020-04-30:
Yup all the things here are very repetitive and SymbiFlow has so many repositories now. I've not used git subtrees before, will try it out.

ryancj14, 2021-05-17:
@mithro This seems like a project we could take on here at BYU. We would start a reasonably new student on it. Our idea is that we will create a new repo that will serve as a container for the items on your checklist, and start our student on adding items to the repo. This project would continue throughout the summer, and anything that isn't done by our student could be added by other SymbiFlow contributors in the future. Our student would probably start by adding the .editorconfig file and some 'documentation around policies' items. Do you think this will be useful? Do you have any other thoughts/direction for us?

ryancj14, 2021-06-09:
@mithro Based on this issue, we have a starting repository located at https://github.com/ryancj14/common-config that could accomplish the goals of this issue. Could you create a common-config repository in Symbiflow where we could put our initial work as a work in progress pull request? It seems like we could get a discussion going in that location for the things we plan to add.

ryancj14, 2021-06-30:
@acomodi @mithro @kgugala The common-config repository has made some progress, and we think that it is the right time to move the repository over to Symbiflow ownership. Thanks to Tim for creating an issue there to help with the progress on that. You can find the repository here at https://github.com/ryancj14/common-config. We are ready to transfer ownership to you. Then we can continue to work on the repo through forks and pull requests. Do you think we should go ahead with that now?