f4pga/ideas #52: Open FPGA for absolute beginners

acomodi, 2020-05-04:
Improve documentation with the following additions:
- explanation of what FPGA is
- how FPGAs work
- what the open toolchain does in a simple form

mithro, 2020-05-04:
A lot of this could come from [the Fomu workshop too](https://workshop.fomu.im).

eine, 2020-06-05:
Ref [Learning VHDL with GHDL (ghdl/ghdl#1291)](https://github.com/ghdl/ghdl/issues/1291)

hackfin, 2020-06-25:
I've made good experience with Jupyter NB and immediate rendering of .dot images or waveforms, however it needs a bit more love (from the Python API side) for better usability.
I've added some beginner's examples to the 'hacker space' which checks out from a secondary repo when starting the binder via the button below (follow instructions)

So basically you code 'HDL' in Python, hit `Run` and see the output from yosys et al.

[image]

The framework also covers co-simulation and synthesis to real hardware (ECP5 FPGAs for now), that's where the focus is currently on. So not sure how well it would integrate with Symbiflow.

josuah, 2022-04-18:
Here is are selected links from [this list](https://josuah.net/links/), in case that helps...

Gateware
--------
**<http://zipcpu.com>:**
A long-winged effort for documenting a learning-path through verilog and hardware design.

**<http://stevenbell.me/>:**
Classes on digital logic circuits available online.

**<https://hdlbits.01xz.net/wiki/Step_one>:**
Exercises and learning material for Verilog.

**<https://projectf.io/>:**
FPGA info site oriented toward video.

**<https://www.fpga4fun.com/>:**
Lessons for practical approach to FPGAs.

HDL Languages
-------------
**<https://en.wikipedia.org/wiki/Verilog>:**
Used by most chip makers, and lingua frinca of the FPGA world.

**<https://en.wikipedia.org/wiki/VHDL>:**
If it is not Verilog, it is likely going to be VHDL.

**<https://clash-lang.org/>:**
Haskell-based HDL.

**<https://m-labs.hk/migen>:**
Python-based HDL.

**<https://www.chisel-lang.org/>:**
Scala-based HDL.

ASIC
----
Specifics of getting an ASIC built-up out of a circuit design.

**<https://www.youtube.com/watch?v=fIJCoViDIj0>:**
Introduction to VLSI manufacturing processes

**<https://hackaday.com/2022/03/18/the-open-source-asics-hack-chat-redefines-possible/>:**
Introduction to the ASIC world.

FPGA
----
**<https://f4pga.org/>:**
fpga toolchain for Xilinx 7-Series, Lattice iCE40, Lattice ECP5
FPGAs, QuickLogic EOS S3

**<https://www.sparkfun.com/news/1203>:**
Introduction to FPGAs by SparkFun

**<https://hackaday.io/project/27550-the-hobbyists-guide-to-fpgas>:**
Introduction to FPGAs published on HACKADAY.

**<https://vhdlwhiz.com/terminology/>:**
Definitions and learning material

**<https://joelw.id.au/FPGA/CheapFPGADevelopmentBoards>:**
Overview and advises of FPGA boards for getting started

Simulation
----------
Simulating hardware at gate or circuit level.

**<https://www.isotel.eu/mixedsim/>:**
Tool building on top of ngspice for simulating circuits.

**<https://en.wikipedia.org/wiki/List_of_HDL_simulators>:**
Wikipedia list of HDL simulators

**<https://ghdl.github.io/ghdl/>:**
The great one tool for VHDL.

**<https://www.veripool.org/verilator/>:**
A simulator turning Verilog into C++, providing a thin C++ api to
interact with simulated hardware.

**<http://iverilog.icarus.com/>:**
Some other simulator that I should try some day!

josuah, 2022-04-18:
Ah, and I forgot, for visual learners: https://redirect.invidious.io/channel/UCJaq8Bzo3J7Hq5lyfhYO3Ew