f4pga/ideas #57: HPC (High Performance Computing) around SymbiFlow stack

manili, 2020-09-15:
Hello everyone,
I'm wondering to know what do you think about a project like SDAccel, but this time around SymbiFlow stack?

Best regards,
Manili

manili, 2020-09-15:
Here is something like what I'm thinking about:
[image]

mithro, 2020-09-15:
You might want to look at what Andre De'Hon and @andrewb1999 are doing around Rosetta;
 * http://ic.ese.upenn.edu/abstracts/prflow_fpt2019.html
 * https://andrewb1999gsoc.blogspot.com/2020/08/gsoc-final-code-submission.html

andrewb1999, 2020-09-15:
@manili Yes, I think a flow like this has a lot of potential and it has been a focus of mine for a bit now.  Andre and I are mostly interested in mapping to separate pr regions within a single FPGA, but I'm sure it's not too difficult to instead partition to multiple FPGAs.

Currently we are still using Vivado HLS while trying to get lower level details in symbiflow working.  Do you have any experience with using Bambu HLS?  We are very interested in switching to an open source HLS tool soon to have a fully open source c to bitstream flow.

mithro, 2020-09-15:
You might also want to explore the newly released XLS from Google (https://github.com/google/xls) too.