f4pga/ideas #59: Add SymbiFlow tooling to openSUSE

mark-stopka, 2020-12-21:
# Brief explanation

[Open Build Service project hardware:FPGA](https://build.opensuse.org/project/show/hardware:FPGA) has been created to accomodate FPGA and ASIC design tools...

We would like to [integrate with webhooks](https://kamarada.github.io/en/2019/03/19/integrating-the-open-build-service-with-github/) for repositories which do not follow conventional release cycles so we have always most up-to-date tools available in openSUSE....

## Expected results

openSUSE will be a usefull Linux distribution for both CI and development of FPGAs and ASICs...

# Detailed Explanation

TBD

## Further reading

* [openSUSE Packaging guidelines](https://en.opensuse.org/openSUSE:Packaging_guidelines)
* [Integrating the Open Build Service with GitHub](https://kamarada.github.io/en/2019/03/19/integrating-the-open-build-service-with-github/)
* [hardware:FPGA OBS project](https://build.opensuse.org/project/show/hardware:FPGA) 

# Knowledge Prerequisites

 * RPM packaging

mithro, 2021-01-20:
@mark-stopka - It appears I never saw this GitHub issue posting last year. A lot of the EDA (FPGA / ASIC / etc) is moving from the SymbiFlow project to https://github.com/hdl

@umarcor is leading an effort to try and support a wide range of packaging styles, including things like native RPMs. He has quite a comprehensive set of documentation at https://docs.google.com/document/d/10_MqFjTIYVVuOJlusJydsp4KOcmrrHk03__7ME5thOI/edit

You might also find it useful to join our IRC channel at #symbiflow on freenode (there is also a slack bridge linked from our website).

umarcor, 2021-01-20:
@mark-stopka, I updated the readme of https://github.com/hdl/packages, so that OBS's hardware:FPGA as well as this issue are shown. Please, have a look and let us know how you envision collaboration areas. The build service integration looks really nice, but it seems to rely on each upstream project configuring a webhook. We don't have permission/authority to do that. Yet, we can communicate and coordinate, for making it easier to achieve.

It is not clear to me what do RPM recipes for OpenSUSE look like and/or where would those be located. Are recipes handled in OpenSUSE's repos (as on Debian, Fedora, Arch...) or are recipes expected to be located in the projects' codebase?