f4pga/ideas #60: SymbiFlow Formal Verification

andrewb1999, 2020-12-23:
# SymbiFlow Formal Verification

Next semester I am doing an independent study on formal verification and am planning on doing a project involving SymbiFlow. My initial thought is performing formal equivalence checking between post-synthesis verilog and the verilog output of fasm2bels. I think this could provide an interesting way for verifying that placement and routing produced reasonable results without having to go back into Vivado.

Would a system like this be useful in the SymbiFlow ecosystem and CI?

I'm also open to other ideas related to formal verification if anyone has one.

mithro, 2020-12-23:
FYI - There is a bunch of documentation in [SymbiFlow Checking / Testing Approach](https://docs.google.com/document/d/11wJUvr2aVBkUiuYYsFN07jkoYr_ccWlLLLFH8YQw8uQ/edit#).

This image should be particularly useful;
[image]

I believe we have some formal verification of things like counters but I'm unsure what the status is.

umarcor, 2021-01-20:
FTR, the state of formal verification using the Property Specification Language (PSL), included in VHDL 2008, and supported by GHDL + Yosys + Symbiyosys: ghdl/ghdl#1616. See also:

- https://github.com/tmeissner/formal_hw_verification
- https://github.com/tmeissner/psl_with_ghdl

In tmeissner/formal_hw_verification, CI is done with containers from [hdl/containers](https://github.com/hdl/containers) (see [docs](https://hdl.github.io/containers)). In fact, recently, @tmeissner helped adding several solvers.

@andrewb1999, feel free to discuss whether the images available in hdl/containers are useful to you, and/or which tools are missing.