f4pga/ideas #67: Deterministic/reproducible builds

walling, 2022-05-10:
I'm new to FPGA development, so forgive me if this idea is unattainable.

Would it be possible to have a FPGA build pipeline that produces deterministic/reproducible builds, i.e. the exact same bitstream for a given Verilog input? This would enable security features, like checking that you deploy the right bitstream and traceability (bitstream hash can be mapped to source).

I guess the question really is whether non-determinism can be removed from all subsystems (synthesis, place & route, etc.) involved in the build process and what the impacts are (performance, resource usage, compile time, etc.).

QuantamHD, 2022-05-10:
Yes, this should be possible. We have reproducible builds in the ASIC space via https://github.com/hdl/bazel_rules_hdl

We would love to add FPGA support to the repo.

walling, 2022-05-12:
@QuantamHD That's interesting. Is it the bazel ruleset that forces the determinism or is it already supported by the individual tools (yosys, nextpnr, etc.)?